PPUC/ZeDMD #2: luminosity percentage

MikePinball, 2022-06-12:
Consider using a percentage rather than a value from 15 to 255 for the luminosity (say 5 to 100 in 5% increments).

Note that the luminosity curve is not a straight line and at the top end you cannot tell the difference in the last 5 values. A table might be better to give real brightness rather than just the value between 0 to 255.

MikePinball, 2022-06-13:
This issue is not important and is only a suggestion.

MikePinball, 2022-06-13:
See here for a discussion on non-linear LED brightness adjustment using a "S" curve. I would use a lookup table with 19 values. I might experiment with this myself and drop a pull request if it works.

zesinger, 2022-06-14:
OK no problem, that's an easy fix anyway. There's no link in the above message.

MikePinball, 2022-06-14:
Link added.

zesinger, 2022-06-14:
Brightness 16 steps (using your link) coded.

closed 2022-06-14 by zesinger