PPUC/ZeDMD #3: Release

MikePinball, 2022-06-12:
Please create a release for the latest binaries and .BAT files. Then you can point everyone to that link for the latest code.

zesinger, 2022-06-13:
OK I'll see how to do that (I'm really noob at GitHub)

MikePinball, 2022-06-14:
Look for "create a release" under the Releases heading on the main page. Then drag and drop your binary released file(s) which could just be a numbered ZIP file. It will automatically also take a snapshot of the source code. Add your own text description. Here is an example from Freezy: https://github.com/freezy/dmd-extensions/releases

zesinger, 2022-06-14:
Thanks for the help.

zesinger, 2022-06-14:
Done here https://github.com/zesinger/ZeDMD_ESP32/releases/tag/ZeDMD

closed 2022-06-14 by zesinger