PPUC/ZeDMD #8: 2.1 for none colored ROMs, seems to be black/white only

Westworld, 2023-01-06:
Thanks for your continuous support and development.
I've updated from 1.2 to 2.1 (none HD display, 128x32).
Running color ROMs works fine.
Running none colored, such as TOTAN 1.4, it normally has 4 level of grey scales, such as full red, 75% red, 50% red , 25% red and off.
This is now running with 100% and 0% only, the values in between are just black.

Going back to 1.2 goes back to working. I missed 2.0 and when downloading the release the binaries are missing, so I cannot compare with 2.0

zesinger, 2023-01-06:
I look at it. Thanks

zesinger, 2023-01-06:
I updated the DMD-ext code, I will release a new DmdDevice.dll in the tuto in a few minutes

closed 2023-01-06 by zesinger

Westworld, 2023-01-07:
The download link in the tuto seems to be invalid?

zesinger, 2023-01-07:
Did you try it in the last 2 hours? It was wrong (due to a copy/past) but now should be OK

Westworld, 2023-01-07:
yes, just now again. Maybe wrong toto? Post #7 in https://www.pincabpassion.net/t15414-comprehensive-tuto-about-colorizingdmd?

zesinger, 2023-01-07:
Oh, sorry! I was on the ZeDMD tutorial, you are right I just updated the link in this tutorial too. Follow this link https://mega.nz/file/iUgTDIBD#qPBDY3Uvs2OT3qKdQGVVvqxKT0hr6H809RvRwrGAnM8