PPUC/ZeDMD #30: ZeDMD reboot after some time with a still image

zesinger, 2023-05-20:
I discovered today what you said to be an intended behavior of the ZeDMD: after some few seconds with a still image, the ZeDMD reboot. The problem is that I use the ZeDMD (and any virtual/real DMD recognized by Freezy's dmdext) in my Serum editor for the "Test Ingame" feature. In this feature, the ZeDMD can keep the same image for minutes, this behavior is not expected at all.
Is it possible to disable it?
Thanks

mkalkbrenner, 2023-05-20:
The feature has been introduced to avoid frozen images on the LCD screen when VPX crashes or doesn't close a table correctly.
We should add a configuration option to control that behaviour.

mkalkbrenner, 2023-05-24:
Version 3.2.3 provides command 26 to turn off that beaviour

closed 2023-05-24 by mkalkbrenner