PPUC/ZeDMD #31: Loss of video output in vpx and PinballY

Scottacus64, 2023-05-21:
I've been having issues with the ZeDMD dropping the DMD display during both vpx and PinballY.  It seems to take about 20 minutes or less and the ZeDMD goes blank, then shows the splash screen for about 15-20 seconds and then goes blank again.  Restarting vpx or PinballY will reestablish the video display on ZeDMD and my third monitor screen (DMD) shows the correct output at all times so it is not a loss of data from PinballY or VPinMAME.  There are no disconnect sounds from Windows so it doesn't seem that Windows has lost connection with the ESP32.  I asked jsm174 who uses a ZeDMD with his mac version of vpx and he says that he gets these drops as well.

Scottacus64, 2023-05-22:
Here's a link to a youtube video of ZeDMD loosing signal to the unit.  It looks to me that dmdExt continues to send signals to the screen DMD.
https://youtu.be/WIeDXoz-5pg

Scottacus64, 2023-05-25:
Thanks for the update, the new splash screens look great!  I ran MM and it froze up after about a minute and a half (video DMD shows active game play but the ZeDMD was frozen on it's last display).  I quite vpx and the ZeDMD still showed the frozen DMD image from MM.  I then started PinballY and ZeDMD showed the PinballY DMD images. After a few minutes PinballY also froze up.  I hope this helps you narrow down the issues.

If you find it easier you can reach me on Discord at scottacus#4075.  The Github mail server won't allow me to attach images of the DMD because of file size.

Scott

v3.2.3 should solve this issue. Could you test it?

mkalkbrenner, 2023-05-26:
DMDext 2.1.2 or newer is required.

Scottacus64, 2023-05-26:
I do have DMDext 2.1.2 installed.  When I can get to the vpin I'll post all of the versions of the various components like VPinMAME so you have a baseline.  Also would there be any value of running ZeDMD in debug mode so I can report the various values when it fails?

mkalkbrenner, 2023-05-26:
The interesting lines in debug mode are 004 and 005. And the command number in the lower right corner.

mkalkbrenner, 2023-05-26:
I tried to reproduce this issue but both, ZeDMD and ZeDMD HD didn't freeze with MM.

Scottacus64, 2023-05-27:
Upgraded to the new 3.2.4 release and grounding out the color change pin (60 second hold) will not put ZeDMD into debug mode.   MM still freezes up for me, please let me know if there are any other file versions of interest to you in troubleshooting.

MM_109 using PIN2DMD.pal dated 11/29/2019
dmdExt bit 64bit v 2.1.2.0
VPinMAME v 3.6 (32 and 64 bit versions)
Pin2DMD libusbk.dll 3.0.7.0
dmdDevice64.dll 2.1.2.0
dmdDevice.dll 2.1.2.0

I have also tried dmdExt 32 bit 2.1.2.0, it seemed to take longer to freeze up (1.5 games).

mkalkbrenner, 2023-05-27:
Setting the device into debug mode only works during frames are sent. So you need a running game.

Are you using PIN2DMD in parallel? Did you try to disable it in DmdDevice.ini?
Any errors in DmdDevice.log after the freeze?

Scottacus64, 2023-05-28:
Thanks for the reply, I hooked my ZeDMD up to my pc laptop and mm runs just fine on that so the problem is with my cab's set up not the ZeDMD or the ESP32 drivers.    

Something is messed up somewhere and I have no idea where it could be. I've checked for dlls in odd locations and there are none so I don't think that is the cause.  Nothing is blocked, and the answer is not obvious at least to me.  I'm now at the point where PinballY won't start if a "real dmd" is enabled so I've made the system worse in the course of trying to figure out the issue.  Which version of Freezy's dmdExt should be used 32 or 64?

I originally built the cab over a decade ago and I'm wondering if I should wipe vpx and do a fresh install of the 32 and add the 64 bit vpx?

zesinger, 2023-05-28:
Perhaps you can try following this https://github.com/vpinball/b2s-backglass/wiki/64bit-VPX-Pincab-Setup-Guide

Scottacus64, 2023-05-28:
Thanks!  I will follow that if I do a reinstall of all of vpx and the associated other software.  Are there going to be registry files that might persist if I uninstall vpx and the associated other programs that might make the problem continue with a fresh install?

zesinger, 2023-05-28:
As the explanation seems to be quite comprehensive, I'd say that there shouldn't be any problem. At least with VPinMame and VPX,  I don't know for the frontend.

Scottacus64, 2023-05-31:
Because my system has been in use for over a decade I decided to do a fresh install of vpx (32 and 64 bit), VPinMAME, FlexDMD and B2S.  I'm happy to report that vpx now drives the ZeDMD without crashes.

As far as PinballY is concerned I've forwarded the following video to Mike showing the behavior of PinballY and the ZeDMD in various "real dmd" settings from "disabled" to "always on".

[https://www.youtube.com/watch?v=J6o_vDY4_i4](url)

It seems to me that:
- PinballY always displays the correct video on the screen dmd
- ZeDMD looses connection to PinballY when a game is started and stopped (real dmd enable always and automatic)
- Adjusting the Real DMD settings will reestablish communication with the ZeDMD

Here's a link to all of the logs and my ini file for dmdExt

[[https://drive.google...iew?usp=sharing](https://drive.google.com/file/d/1EzaEMYshuyuwYCMvE_ANqoKyNv4MK-D-/view?usp=sharing)](url)

Scottacus64, 2023-06-02:
I figured out what the problem was and it was on my end.  Back in 2019 Mike wrote a javascript for me that displayed information on the DMD for games that didn't have an nvram.  This was before he incorporated the PinEMhi into PinballY and it has been sitting in my system since then.  This was the cause of all of the issues and once that main.js was removed, all of the ZeDMD issues were resolved in PinballY. 

Sorry for any inconvenience these error reports might have caused!

closed 2023-06-02 by Scottacus64