PPUC/ZeDMD #34: Gamma correction

mkalkbrenner, 2023-05-27:
Some games produce rather dark DMD outputs. Instead of just increasing the brightness it would be good to perform a gamma correction on demand.

mkalkbrenner, 2024-01-29:
Obviously the HUB75 lib already does some gamma correction

PastorL69, 2024-11-08:
I analyzed the current low brightness pixel colors with RGB565 and the cutoffs on my ZeDMD.
Stern SAM has the most brightness possibilities, which is 16 (including one that is entirely dark, so OFF). We should stick to that and base everything off of that. 

So, the first pixel brightness starts at 6% out of 100, then 13%, 19%, 27% etc. Currently this 6% doesn't display at all on my ZeDMD. But, this value should definitely be displayed at all costs. So essentially what I am saying, the cut off for the pixel NOT to be displayed is a value below 6%. At this point the pixel can be regarded as black, this is too dark, so the ZeDMD not displaying a pixel that is below the value of 6% is perfectly, we should not touch that for sure. 
I attached a picture with the RGB values of white/gray color at 6%. 
[image]

Ok, now that we know that, here is my suggestion to fix this issue: 
I've found that raising every color possible to 10% should display every pixel possible. these are the values for a gray/white pixel: 
[image]

So, every pixel that is between 6% and 10% should be raised to 10%. This could be a nice first test.

PastorL69, 2024-11-17:
https://www.youtube.com/shorts/OsuDI9w9BiQ

Testing shows that a 1.3x brightness increase in the entire palette works the best. 

As said in the post above, before the 6% brightness pixel would not show up, now it does. 
So, with the 130%/1.3x applied it solves the current gamma issues, at least for the Stern SAM games I have been doing. Not only does it fix the 6% brightness pixel issue, it also improves the visibility of the imported video scenes, as certain pixels that were below the displaying threshold before are now no longer.

PastorL69, 2024-11-22:
With the transition to 8bit this one can be closed.

closed 2024-12-09 by mkalkbrenner