PPUC/ZeDMD #39: zedmd various problems

Pinbalbilly, 2023-07-14:
Hi, I don't know if this is the right place to report bugs I'm having with zedmd.  I have Vpinmame 3.6, frezzy 2.2.0 beta2, in zedmd HD, I find the following errors: -sega tables Baywatch, maverick, Frankenstein, the dmd image is not understandable, everything is mixed up, it is impossible to use it.  - tables data east hook, Batman, checkpoint, turtles, the dmd comes out double, duplicate, attached photos.  Any help to fix the problem?

Sega

[image]

Data East

[image]

mkalkbrenner, 2023-07-23:
I just tested Baywatch and Star Trek with ZeDMD HD, both work perfectly well with DMD Ext 2.1.2.

But I'll try the dev version DMD Ext now.

Pinbalbilly, 2023-07-24:
> 

ok, I wait for news, thanks

mkalkbrenner, 2023-12-09:
Will be fixed by https://github.com/freezy/dmd-extensions/pull/450

closed 2023-12-09 by mkalkbrenner