PPUC/ZeDMD #43: ZeDMD double screen ? (128x16)

aerao, 2023-07-23:
I'm currently having a problem with my ZEDMD, again and again on the Star Trek 25th generation table (128x16). Everything would work correctly since the last updates of the ESP32 chip. I don't know if it's linked to the 64-bit version (VPX, frenzy dll v2.2.0-BETA2 etc...), but nothing works properly anymore, I now have a double screen, only on this table.  No problem on the others.
Here's a visual to give you an idea of the problem. I've tried modifying my dmdDevice.ini file in every direction, but it doesn't change anything.
Any ideas?

https://youtu.be/mYHx7sZRRYE

mkalkbrenner, 2023-07-30:
I now tested ZeDMD with Baywatch and Star Trek. Both modes work perfectly well with ZeDMD (normal and HD) and the latest dev version (master) of DMDext in 64bit!
According to github, the latest version in master has no code changes compared to v2.2.0-BETA2

mkalkbrenner, 2023-12-09:
Will be fixed with https://github.com/freezy/dmd-extensions/pull/450

closed 2023-12-09 by mkalkbrenner