PPUC/ZeDMD #44: I have a problem with Zedmd if I use an altcolor

lglinux3d, 2023-10-09:
Bonjour, 
j'ai un soucis avec Zedmd si j'utilise un altcolor .pal etc... lorsque je sort de VPX ou a la sortie de table de pinup celui ci reste affiché sur Zedmd
Existe il une commande ou une solution pour faire disparaitre l'affichage enfin éteindre les panneaux leds ;)
Hello, 
I have a problem with Zedmd if I use an altcolor .pal etc... when I exit VPX or pinup table it remains displayed on Zedmd
Is there a command or a solution to make the display disappear or turn off the led panels ;)
Thks for you reply ;)

mkalkbrenner, 2023-12-09:
Maybe will be fixed by https://github.com/freezy/dmd-extensions/pull/450

mkalkbrenner, 2024-03-11:
https://github.com/PPUC/dmd-extensions/releases/tag/v2.2.2-zedmd.rc.1