PPUC/ZeDMD #45: No Ready Signal

francisdb, 2023-11-06:
Trying to get ZeDMD running on macOS / Linux with vpx standalone but it looks like the dmd is never ready.

ZeDMD `3.2.4`

Any hints on how to debug this?

```
2023-11-06 14:49:14.470 INFO  [14383] [ZeDMDUtil::OnLogMessage@12] ZeDMD found: device=/dev/ttyUSB0, width=128, height=32
2023-11-06 14:49:14.470 INFO  [14393] [ZeDMDUtil::OnLogMessage@12] ZeDMD run thread starting
2023-11-06 14:49:14.478 INFO  [14393] [ZeDMDUtil::OnLogMessage@12] No Ready Signal
2023-11-06 14:49:14.486 INFO  [14393] [ZeDMDUtil::OnLogMessage@12] No Ready Signal
2023-11-06 14:49:16.888 INFO  [14393] [ZeDMDUtil::OnLogMessage@12] No Ready Signal
2023-11-06 14:49:18.905 INFO  [14393] [ZeDMDUtil::OnLogMessage@12] No Ready Signal
2023-11-06 14:49:20.921 INFO  [14393] [ZeDMDUtil::OnLogMessage@12] No Ready Signal
```

jsm174, 2023-11-06:
I just tested with 3.2.5 and it is working fine. Its been a while but I think the protocol switched between 3.2.4 and 3.2.5.

francisdb, 2023-11-06:
Build instructions to end up with `3.3.0` master
macOS, as on Linux (Ubuntu 23.10) I got some python errors

```bash
git clone git@github.com:PPUC/ZeDMD.git
cd ZeDMD
brew install platformio
pio run -t upload
pio run --target buildfs
pio run --target uploadfs
```

francisdb, 2023-11-06:
And while I am documenting things you also need this on Ubuntu 23.10 so your user can acces `/dev/ttyUSB#`

```
sudo gpasswd --add ${USER} dialout
```

jsm174, 2023-11-06:
Cool. I looked in my history and I have a:

`sudo chmod 666 /dev/ttyUSB0`

mkalkbrenner, 2023-12-09:
Does VPX Standalone work well now with libzedmd 0.1.0 and ZeDMD 3.3.0?

francisdb, 2023-12-09:
Yes, there are some remaining issues but this one is fixed. Thanks.

closed 2023-12-09 by francisdb