PPUC/ZeDMD #52: ZeDMD displays "Unsupported render mode: #"

ZiggyDaMoe, 2024-03-06:
I run VPinballX_GL64.exe 10.8.1873. I also tried VPinballX107_32bit.exe 10.7.404 Final. 
Every table I run works correctly and with altcolor, "Use External DMD (.dll)" is checked. I also tried disabling color and just had "Use External DMD (.dll)" checked in PinMAME 3.6
DmdDevice64.dll 2.2.1.0 from Freezy
DmdDevice.ini set:
[zedmd]
; if false, doesn't bother looking for a ZeDMD
enabled = true
 All I get on the external PinDMD is "Unsupported render mode: 8 and then the number switches to 9 with a .vni and .pal altcolor. When I try an altcolor with .cRZ I get "Unsupported render mode: 11".  I have also gotten "Unsupported render mode: 4".
I also got "Unsupported render mode: 20" when I switched to another DmdDevice64.dll from https://github.com/PPUC/dmd-extensions/releases/tag/v2.2.2-zedmd.beta.3

I can see the ZeDMD in my Device Manager Win 11
[image]

When I run a test I got this information:

dmdext test -d auto
 [1] 2024/03/05 22:36:07.020  WARN | Device PinDMDv1 is not available.
 [1] 2024/03/05 22:36:07.090  WARN | Device PinDMDv2 is not available.
 [1] 2024/03/05 22:36:07.363 ERROR | Error: The operation has timed out.
 [1] 2024/03/05 22:36:07.474  WARN | Device PinDMDv3 is not available.
 [1] 2024/03/05 22:36:07.918  WARN | Device PIN2DMD is not available.
 [1] 2024/03/05 22:36:07.926  WARN | Device PIN2DMDXL is not available.
 [1] 2024/03/05 22:36:07.926  WARN | Device PIN2DMDHD is not available.
 [1] 2024/03/05 22:36:08.043 ERROR | Error: Access to the port 'COM3' is denied.
 [1] 2024/03/05 22:36:08.043  WARN | Device Pixelcade is not available.

I can provide additional information if there can be anything I should be looking into.

mkalkbrenner, 2024-03-11:
As mentioned at VPU, you don't use a ZeDMD, but a PINDMDv3 running the (potentially modified) ZeDMD firmware.
WHere did you download the firmware and how did you update your device?

ZiggyDaMoe, 2024-03-28:
Thank You for all the great information. From what I have been able to obtain suggests that the PinDMDv3.1 is using hardware that follows all of the suggested build information I found from:
https://www.pincabpassion.net/t14796-zedmd-installation-english
I updated the Firmware to v3.5.2 from the ZeDMD Updater application which should also be the straight forward standard method.
It seems that it is built on an ESP32 connected to a USB UART (UART is a Silicon Labs CP2102N)
I was intimidated to initiate a project like the ZeDMD. I knew there was a steep learning curve for me.
Now that I have the PinDMDv3.1 and have investigated more of what is involved in building a "ZeDMD" had I started from scratch I feel like it would be nearly identical to the PinDMDv3.1 just not as expensive. Sometimes education costs you a little bit of money.

mkalkbrenner, 2024-05-03:
Use firmware 3.6.0 and DMDextensions 2.2.2.
ZeDMD and PINDMDv3.1 support has been improved a lot in DMDext 2.2.2.

closed 2024-05-03 by mkalkbrenner