PPUC/ZeDMD #54: Compilation instructions

Jay-Zil, 2024-05-02:
Hello,

I have built the 128x32 environment with VSCode and PlatformIO as extension and I get a firmware.bin that is 327kB in size compared with ZeDMD.bin downloaded in last release which is 4096kB.

Can you tell me how to build the firmware please?

Best regards,
Jay.

mkalkbrenner, 2024-05-03:
The firmware consists of two parts:
1. the "code"
2. the file system

You can directly build and flash the file system and the code form VSCode using the Platformio plugin.

The 4096 kb binary is a complete image that contains both. It is a bit more complicated to build:
https://github.com/PPUC/ZeDMD/blob/ccb721335ea2bf4ba8b8b00375ca2e987f26ddb0/.github/workflows/ZeDMD.yml#L57

Jay-Zil, 2024-05-03:
Thank you for your reply. I'm glad to learn how it works.
I managed to compile the code, merge and transfer it.
Thank you so much.

closed 2024-05-03 by Jay-Zil