PPUC/ZeDMD #55: ZeDMD turn off in pinballx when close table and open another.

strangeleo7, 2024-05-31:
Hi, how are you?
My name is Leo, I live in Argentina and I bought a ZEDMD in ULTRAPIN, it arrived with version 3.5.2
I already have it working CONGRATULATIONS IT IS WONDERFUL!
Now I have some doubts...
When I leave a table in pinballx and want to re-enter it turns off (3.5.2 is on the side) it tries to start and it turns off. If I reset it and enter another table it enters fine. What could that be?
And on the other hand, is there a way to show me something in pinballx? THANK YOU in advance, I hope for a prompt response THANK YOU!

mkalkbrenner, 2024-05-31:
We think that this is an issue with DMDext at the moment. This is the corresponding issue:
https://github.com/freezy/dmd-extensions/issues/481

strangeleo7, 2024-05-31:
THANS MARKUS! I sent you several messages by different means, thank you for answering me by EMAIL!! Today when I get home I try it... I hope that with 2.2.3 it will be solved.. Is there a way for the ZEDMD sign to appear every time you exit a game, and not go to number 2.5.3 (very thin at the bottom left)?

 

freezy/dmd-extensions#481 <https://github.com/freezy/dmd-extensions/issues/481>

mkalkbrenner, 2024-10-31:
fixed in dmdext:
should be fixed with the commit of [#505](https://github.com/freezy/dmd-extensions/pull/505).