PPUC/ZeDMD #63: try to load in esp32 s3

shiryu55, 2024-10-02:
supposedly loaded the firmware corresponding to my s3, ZeDMD-S3-N16R8_256x64 by this command line.
esptool --chip esp32s3 write_flash 0x0 ZeDMD.bin
I finished the firmware loading process correctly.

no response from the panels

mkalkbrenner, 2024-10-04:
Are the wires correct? External Power supply for the panels?

shiryu55, 2024-10-04:
[image]

I have it connected this way, plus the pins on the panel with a PC power supply
However, with the normal ESP32 programmed with the app, the panels work perfectly, so it is not a power problem.
It could be a firmware loading mode problem on the esp32 s3. I'm out of ideas.

shiryu55, 2024-10-04:
[image]
As you can see in this other screenshot, I have no problem with the old esp32. 
perform firmware upload with ZeDMD_Updater.

It's something related to the S3 firmware or the way to load it.

mkalkbrenner, 2024-10-04:
Are you sure that your S3 is a N16R8?

shiryu55, 2024-10-04:
[image]

shiryu55, 2024-10-04:
0532042018040028018160004138ESP0V203

shiryu55, 2024-10-04:
[image]

shiryu55, 2024-10-05:
I don't know if you could try your compilation  https://github.com/PPUC/ZeDMD/releases/download/v3.6.0/ZeDMD-S3-N16R8_256x64.zip with an S3.
or some step-by-step guide on how to compile it myself
I tried to compile the source version 3.6 , platformio.ini modify [platformio]
default_envs = S3-N16R8_256x64
but it generates a 300kb firm, I don't know what I'm doing wrong

mkalkbrenner, 2024-10-09:
try to build from 'main' branch, not 3.6.0

mkalkbrenner, 2024-10-09:
Or flash this build:
https://gitlab-ee.intern.jarltech.de/cms/shop/-/merge_requests/2908

shiryu55, 2024-10-09:
compile the main version and then 3.6 with the same result

shiryu55, 2024-10-09:
[image]

shiryu55, 2024-10-09:
the link is not right 
Anyway, could you try the firm 3.60 that you compiled for the s3 and can you confirm how it works in case I make a mistake or is it the compiled file?

shiryu55, 2024-10-11:
a build guide would be great

shiryu55, 2024-10-12:
Recompile the main version with the new changes, I'm still the same doesn't show any signs of life the panels.

gustavoalara, 2024-10-14:
Hi @mkalkbrenner 

Same for me, I'm trying to load the firmware to the same S3 Shiryu55 uses. The result are the same. No problem uploading using the right USB conector on the S3 but when connecting to the left connector the panel doesn't show anything and the ESP32 continuosly connects and disconnects on Windows.

mkalkbrenner, 2024-10-15:
S3 support is still experimental and has known issues, especially with USB. If you want to test it, you should build and flash the latest version of the main branch. That could be done from VSCode with platformio plugin or via the platformio command line tool. Or download the latest github action artifact and flash it using the the esptool:
https://github.com/PPUC/ZeDMD/actions/runs/11259693865
If you don't know how to do that, it might be better to wait for a release.

shiryu55, 2024-11-08:
I recently saw the new revision, 4.0.  I loaded it for now on my S3 without connecting to the panel, how could I check the configuration web interface from the PC? to see how it works.

mkalkbrenner, 2024-11-08:
we had an issue with the bootloader in the bin files for S3. Should be fixed now.

riyadchowdhury, 2024-11-10:
Btw in the above screenshot C pin is connected to GPIO46 but if you look at https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/src/platforms/esp32s3/esp32s3-default-pins.hpp C pin is actually on GPIO 3. I was having issues until I connected 3 to GPIO3 on a esp32-s3

shiryu55, 2024-11-10:
Steps followed, pin by pin with Dupont following this diagram
[image]

up the updater to 4.8.1
loaded with the command

esptool --chip esp32s3 write_flash 0x0 ZeDMD.bin

[image]

the panels are not responding”

shiryu55, 2024-11-10:
“Hello Markus,

I’ve tested the new firmware version (4.0.2) with the ESP32, and everything is perfect; the improvement in smoothness is very noticeable.

However, with the S3, I’m still having issues and can’t get anything to display on the matrices.

These are the steps I’m following:
	1.	This is the pin assignment I’ve set up (photo attached).
	2.	I connect the USB to the left port of the S3 and load the firmware using the latest version of esptool (4.8.1)
	3.	The firmware seems to load correctly (screenshot attached).
	4.	I connect the USB to the right port of the S3. But nothing is displayed on the screen.”

[image][image]
[image]

mkalkbrenner, 2024-12-09:
Meanwhile we tested so much with different variations of the S3.
Ensure that you use libzedmd 0.7.6 and ZeDMD 4.1.0.

shiryu55, 2024-12-10:
Hello, could you detail the step-by-step process for uploading to S3 in case I make any unusual mistakes?

As for libzedmd, I can’t test it because no information appears on the panels, as I explained to you earlier.

shiryu55, 2025-01-06:
Hello Markus, after countless attempts and tests, it turns out there was a cold solder joint on the OE pin. Everything is now working correctly.

mkalkbrenner, 2025-01-06:
Thanks for your feedback!

closed 2025-01-06 by mkalkbrenner