PPUC/ZeDMD #64: frame missing on some VPX Table

CfiroteCode, 2024-10-20:
Hi everyone,

I have some frame missing on some very detail (full color) DMD 

Exemple of table : 
JP's Deadpool v5.1.2_VPX8
SpongeBob's Bikini Bottom Pinball VPW 2.1

they are full color and animated DMD, on my computer the framerate is ok, on my ZEDMD ESP32, there is low framerate and some ar juste not showing.

can i lower the color resolution (to 256 by exemple) ou can i setup a frameskip ?

I've also see on a pull request a SPI communication, does the limitation is about the com port data rate ?

I use the 3.6.0 version of ZEDMD with a USB cable.

mkalkbrenner, 2024-10-25:
The USB to serial converter used by the ESP32 is simply to slow.
Frame skipping is already built-in.

Ths SPI communication targets a setup in real pinball machines.

To solve the issue for the virtual pinball, we want to switch to the ESP32 S3 and native USB communication. But it is still under development.

The other option is the WiFi mode. The communication speed has been improved recently and ZeDMD 4.0.0 should be released "soon".

mkalkbrenner, 2024-12-09:
ZeDMD 4.1.0 will contain another performance boost.

closed 2024-12-09 by mkalkbrenner