PPUC/ZeDMD #67: Feature request: Automatic reset (after timeout)

jueank, 2024-11-17:
In practice, it happens often that when leaving a table, the content on the DMD is still being shown. The reasons for this might be different: Improper scripting in VPX, VPX crashing or being terminated externally, etc.
While being in the Frontend - or moreover - when playing another non ROM-based table, this content stays visible on screen. This is a bit disturbing.
It would be nice, if the DMD is being reset to black by some criteria. Either a configurable timeout (x seconds) - this should be sufficient, as there are usually very frequent updates to the DMD while playing. Alternatively it could be detected if there is an active connection over the serial port.

mkalkbrenner, 2024-11-17:
I think that ZeDMD is the wrong point to implement that.
That issue  happens for every DMD.
dmdserver, the Dmdext equivalent for other operating systems than windows already handles such things.

jueank, 2024-11-18:
Thanks for the clarification. I will rephrase my feature request and raise it in the Dmdext project.

jueank, 2024-12-17:
Hi,
I have to come back to this feature request. 
I think, one of the typical cases is that ZeDMD is losing the serial connection to Dmdext. For example by 'hard' terminating a table, or a crash, or incorrectly written table scripts. Then Dmdext has no chance to take care of resetting ZeDMD or setting DMD content to black.
Only ZeDMD itself can reset itself to a black screen or the initial screen showing the version number in the lower corner.
Losing the serial connection (plus a timeout) could be exactly the trigger to initialize that.

zesinger, 2024-12-18:
Have you tried the Wifi version?

commit d5fb7fb7 (2025-01-28): display logo after timeout, closes #67
closed 2025-01-28 by mkalkbrenner