PPUC/ZeDMD #69: LilygoS3Amoled/ZeDMD-Wifi corrupted after changing values in built in website

jsm174, 2024-12-02:
After flashing the latest ZeDMD-Wifi on a LilygoS3Amoled, if you visit the built in website and change a value such as brightness or dmd style, the device becomes completely non-responsive. Unplugging and reconnecting doesn't help as the display is completely dead.

The only solution is to put the device back into programming mode, completely flash the firmware AND filesystem. After the filesystem is restored, unplugging and reconnecting the device brings the display back to life.

I'm not the only user who has experienced the behavior, so figured best to create an issue.

zzmadzz36, 2024-12-03:
Yeap had the exact same issue as jsm174. Had to reflash the firmware and the file system. 

All I did to brick the lilygo board was to access the configuration website, modify the wifi settings (which didn’t work) and changing some of the brightness settings and then the entire screen turns black and refuses to do anything.

mkalkbrenner, 2024-12-09:
I don't think that this issue exists anymore.
Test again with latest dev version.

mkalkbrenner, 2025-01-29:
I can't reproduce it