PPUC/ZeDMD #73: ZeDMD: using one 64x32 panel

sic79, 2024-12-15:
Would it be possible to add support for only one 64x32 panel?

mkalkbrenner, 2025-01-06:
What should it display?

sic79, 2025-01-11:
Hi @mkalkbrenner, I was thinking about a project for my Arcade Cab showing some playerstats on a 64x32 DMD. But as for today, I think I actually will go with 128x32 instead. So if there is no other than me who is interested in this you can close this "issue" for now, thanks :).

mkalkbrenner, 2025-01-13:
I just wondered if you want to downscale pinball ROMs.

Let's postpone this one after ZeDMD 5.0.0

sic79, 2025-01-14:
Great, and thanks for making ZeDMD. Works very well on my setup👍

mkalkbrenner, 2025-02-04:
As a workaround you could use he 128x32 firmware and send 128x32 content. I this case only the left part will be displayed.