PPUC/ZeDMD #75: Mapping cables for esp32 wrover or esp32 S3 N16R8

Sylvain-Kraisin, 2025-01-09:
Hello. First, thank you for your great work. 

I'm using it on my batocera setup (ZEdmd over usb with a esp32 wroom) and it s so cool.

The problem is on full animated gifs, it is not smooth so i tried to use an esp32 wrover (with ZEdmd over usb) and also an esp32 S3 N16R8 (with ZeDMD S3 over usb) but i cant figure out how i should plug the pins. 

There is a tuto with picture for the classic esp32 wroom but i found nothing for the 2 others esp32. 

I tried myself to redefine GPIO16 and 17 on the wroover since they are used for ram, but even i have the source code, i guess i m not competent enough. 

May you have answers for helping me and others who might want use these esp32 for more smooth animations ? Regards.

mkalkbrenner, 2025-01-09:
S3 Support is still experimental in ZeDMD 4.x.

The upcoming ZeDMD 5.0.0 will be faster but also requires changes to libzedmd and therefore batocera.

I expect ZeDMD 5.0.0 and libzedmd 0.8.0 to be released within the next weeks.

https://youtu.be/hD6v0Xifu8I

mkalkbrenner, 2025-01-09:
The documentation for the ESP32 and the S3 will be written based on 5.0.0

Sylvain-Kraisin, 2025-01-09:
> The documentation for the ESP32 and the S3 will be written based on 5.0.0

Thank you sir. Can't wait for the 5 !

So you mean there will be a tutorial for link pins from esp32 wrover and esp32 S3 to the led cable like it is today for esp32 wroom but it will need an update in Batocera code in order to work ?

Wesie, 2025-01-14:
Otherwise you can find earlier the assigned pins in the definition of following source file:

https://github.com/PPUC/ZeDMD/blob/main/src/displays/LEDMatrix.h

closed 2025-01-27 by mkalkbrenner