PPUC/ZeDMD #77: help ver 5.0 and s3

shiryu55, 2025-01-18:
I recently uploaded version 5.
After uploading the firmware from the command line, a selection menu appeared on the panels. Then, I checked the code and added this Arduino pad to select the options. I exit the menu through “Exit” and notice the loading bar to the right, next to the version 5.0 logo, but it stays there, and I can’t get it to display anything on the panels.

shiryu55, 2025-01-18:
[image]

This is the pad I used.

mkalkbrenner, 2025-01-18:
did you update libzedmd to 0.8?

shiryu55, 2025-01-19:
yes , https://github.com/PPUC/libzedmd/releases/tag/v0.8.1

[image]

[image]

I tried running it this way,
but it gives me errors.

With version 4.0.2, I didn’t have these issues.

Wesie, 2025-01-19:
In both pic`s in the log are 0.8.0 written... Is this an internal mismatch of the dll or really the lib 0.8.0 instead of 0.8.1?

shiryu55, 2025-01-19:
[image]

Test performed with the libzedmd win x64 version 8.1.

shiryu55, 2025-01-19:
The DMD turns off for one second and then returns to the screen with the ZEDMD logo in standby mode.

shiryu55, 2025-01-19:
I’ve noticed something else: with version 4.0.2, the image was more stable when recording with a phone. At first glance, the frequency issue isn’t noticeable. I’m sharing two short videos—one with the menu from version 5.0 and another with an animation from version 4.0.
The S3 connects to the panels via USB. Regarding version 5.0, I am unable to load any animations, so I cannot provide an animation video.

menu ver 5.0
https://github.com/user-attachments/assets/5efced86-3c4a-43d1-83ad-b310835eeb80 

vídeo versión 4.0.2
https://github.com/user-attachments/assets/dfaacc5d-f8e6-4657-9325-48816e07ac74

PastorL69, 2025-01-22:
Version 5.0.0 is still being worked on. 
USB CDC for Windows is one of those things, so it's expected that you do not get a datastream for your "menu ver 5.0" video.

[image]

It states that S3 5.0.0 with USB is not yet implemented. I don't get the point of this issue, the feature doesn't exist yet.

closed 2025-01-23 by shiryu55