PPUC/ZeDMD #83: Mini-z error -3 with USB

RED31FR, 2025-01-30:
First thanks a lot for all your wonderfull works!! 

I used you ZeDMD firmware and libzedmd dll with my application. All these things work perfectly with version 4.0.2 and below.

With the last version of lib and firmware I have mini-z error -3 quickly after some frame update and the connexion to the ZeDMD is lost.

Could you help me on this?

Thanks

mkalkbrenner, 2025-01-30:
Please complete the following information:
 - OS: [e.g. Windows 10, macOS 15.2, Ubuntu 24.04, batocera 41, iOS, Andorid, ...]
 - Client [e.g. DMD Extensions, dmdutils, VPX Standalone, Popper, Pinball Y, ...]
 - ZeDMD Version [e.g. 5.1.1]
 - libzedmd Version [if known]
 - ESP32 variant [e.g. ESP32, S3, S3 Amoled, ...]
 - Panel Size [e.g. 128x32]

RED31FR, 2025-01-30:
Thanks for your quick message, my answers below ;-)

OS: Windows 11
Client: Mine :-), I tried zedmd_test.exe too, some time work, some time not.
ZeDMD 4.0.2 work great!
ZeDMD 5.1.1 miniz error -3 quickly appear
libzedmd: v0.75 for ZeDMD 4.0.2
libzedmd: v0.90 for ZeDMD 5.1.1
ESP32 WROOM 32
Panel size 256X64 with smallcab shield

BR

RED31FR, 2025-02-04:
Thanks for your last update firmware/lib, all works well!

closed 2025-02-04 by RED31FR